Mattiwatti/EfiGuard #91: Ventoy support

acoul, 2023-05-26:
Can EfiGuard be used with Ventoy ? I tried to create an iso out of a USB key, but I failed :(

Mattiwatti, 2023-09-25:
I just tried this, and the short version is that no, it doesn't look like it can. The reason for this is the fact that EfiGuard is unloading itself when you are booting into Ventoy instead of booting the Windows Boot Manager directly, because it thinks you're booting something other than Windows. Well, you **are** booting something other than Windows, it just so happens that Ventoy can *also* chainboot Windows, which I agree, EfiGuard should support.

So the good news is that I know what the issue is and I can probably fix it, but the bad news is that the naive fix (simply not unloading the driver) leads to other potential issues, for instance: the sequence
1. EfiGuard
2. Ventoy or similar boot manager
3. Linux

breaks with this fix, and in a worse way, because Linux does not even boot. This surprises me a little, since I don't recall ever having an urgent **need** to unload the DXE driver when not booting Windows, I just thought it would be neater this way. So I will have to investigate some more to figure out a fix for this that works with all operating systems.

Mattiwatti, 2023-09-26:
9aff366 should fix this. Thanks for reporting.

commit 9aff3669 (2023-09-26): EfiGuardDxe: delay driver unload when a non-Windows OS is booted

Fixes #91
closed 2023-09-26 by Mattiwatti