Mattiwatti/EfiGuard #93: Multiboot

acoul, 2023-05-30:
I have a system with two internal SSDs, one has windows 8 & the other-one windows 7. booting EfiGuard from USB it always defaults on the first system it finds. is there a way for EfiGuard to detect the different systems & have a prompt for specifying which system to boot ?

MrTatiTunes, 2023-06-01:
Hi Acoul we are meeting again, here it looks like I have problems with booting multi disk multi system, I hope we will finally solve our troubles.

acoul, 2023-06-01:
greetings again,

FWIW, I got a workaround for my issue. 

I did rename on the NVME on the EFI/FAT16 partition the EFI\Boot\bootx64.efi to EFI\Boot\bootx64.bak.efi & EfiGuard on the usb key jumped to my external usb win8 system

so for me this is now a feature request rather a bug issue. a menu selection on multiple windows systems found would be handy, but the above workaround works

Mattiwatti, 2023-09-26:
Hi,

I'm not planning to add a boot selection menu to EfiGuard's loader at the moment, mainly because this is something that would be quite complex to actually do correctly and in a way that covers all use cases. The entire loader app consists of a single C file. Just like the user mode executable, the loader is mostly just a helper tool to make the driver do things. It is definitely not a full blown boot manager like rEFInd or Clover and I'm not planning to make it one.

Reading your description, I do think that what you're trying to do should be possible in some other ways:
1. Using EfiGuard without the loader app: if you can boot into the UEFI shell, you can add the DXE driver as a `Driver####`UEFI variable using `bcfg`. To do this you should copy the driver file to the system ESP (not a USB stick) and run in the UEFI shell:
  `bcfg driver add 00 EfiGuardDxe.efi "EfiGuardDxe"`
  Assuming your firmware supports this correctly, EfiGuard should now be automatically loaded from the ESP on every boot. NOTE: I've seen systems on which you have to run this command with add**p** instead of add, and some on which you have to run it with add instead of add**p**. If it doesn't work with the first command, try again with `addp`.
  Why do I think doing this will help you here? Because using this setup you no longer need the loader app, which as you've noticed is fairly primitive and not really meant to support interactive boot selection. Instead you should be able to simply use your motherboard firmware's boot selection menu to choose either Windows 7 or 8, and EfiGuard will still be loaded automatically no matter which you choose.
2. Another possible setup I can think of would be to have a "shared" BCD store in one of the two Windows installations, containing BCD bootloader entries for both Windows 7 on one disk and for Windows 8 on the other. You can then make this Windows installation the "chosen one" for the EfiGuard loader by making it the second boot entry after it in the BIOS. After EfiGuard loads the Windows boot manager you can then make your choice (Windows 7 or 8) at the Windows boot manager selection screen instead.
  EfiGuard should work with both of these, meaning it should work even if the Windows installation being booted is on a different drive than the one containing the Windows boot manager. I do this kind of "BCD sharing" pretty often myself, because it's quite annoying to have to maintain a separate BCD store for each drive that contains a Windows installation.

closed 2023-09-26 by Mattiwatti

acoul, 2023-10-04:
thank you so much for your answer and details (which are always) of great value.

reading your elaborate input is like going back to the (good old) university days (really)

best regards

acoul, 2024-06-12:
just a quick follow-up on this. I mainly use EfiGuard on my portable usb toolkit disk, thereof a multi-boot feature is a must for my needs

my approach for loading EfiGuardDxe.efi as a device driver is to use [ventoy](https://github.com/ventoy/Ventoy) along with [GrubFM](https://github.com/a1ive/grub2-filemanager)

when inside GrubFM you may easily navigate and first load EfiGuardDxe.efi as device driver and then navigate and load the desired OS