Mattiwatti/EfiGuard #94: It looks like I have problems with booting multi disk multi system.

MrTatiTunes, 2023-06-01:
The story is like this:
I moved 4 systems from Z97x gigabyte cloning it trough network to new MVMe M.2 and finally managed them to update and settle in new environment. 
But the Compatible and UEFI start is not OK.
I have ghost screens of choice of ghost systems, and there are up to 3 of them, despite I choose only start from M.2 in BIOS.
How to trace and delete all BCD options and have the only one, I wish to start from UEFI, but it not let me, all the times, sometimes yes sometimes no. it is not easy to remove all drives from 3 slot M.2 system and some SATA drives.
Please help to do it one time for all.
A bit tired with chasing start options :)

Link to film from pc start:
https://youtu.be/zKA_cRkyosM

MrTatiTunes, 2023-06-01:
Just checked the editor you named,
finally something is showing nests of EFI data :) maybe this will help me, finally, thank you so much.
I used many of UEFI/EFI/BCD editors and none of them have been starting with positions of EFI data...
Now doing backup-ing of partitions, moving and reducing hardware inside PC. This should help also.

MrTatiTunes, 2023-06-02:
> you can try using something like an [EFI Boot Editor](https://www.majorgeeks.com/files/details/efi_boot_editor.html) (I haven't tried this myself) & see if that helps cleanup & organize your UEFI boot menu.<
This EFI editor shows something, but not enough to solve surely the issue.

Have you knowledge to explain how to start this EfiGuard tool in PC Windows environment?
Z590 and intel 11 system is not easy starting from USB :) showing all HDD and NVME storages to manipulate the BOOT choices. Seems like single drive single systems are preferred by devs even here, Mattiwatti refused to explain those questions nor give more clear/detail instruction.

Mattiwatti, 2023-10-03:
>Mattiwatti refused to explain those questions nor give more clear/detail instruction

(posted **1 day** after the issue was created!)

I think a more likely explanation might be that I simply hadn't gotten around to doing so yet, because while I do generally try to reply to "question" issues, I don't love them. Especially when, as @acoul already pointed out, the questions are not even related to EfiGuard at all. Consider that if I had seen this issue sooner, it would have *already* been closed.

>Seems like single drive single systems are preferred by devs even here

I think you'll find that actually, you're wrong. I have so many drives, and partitions on these, in my development machine that I am very nearly out of DOS drive letters to use. And that is just one machine. All of my systems have multiple drives, and all of them have at least 3 different operating systems installed. Consider that a more likely explanation is that you're simply doing something wrong.

You should learn the difference between firmware (UEFI) boot entries and BCD entries:
1. UEFI boot entries are stored in NVRAM and contain entries for boot managers. Common UEFI boot managers: GRUB, rEFInd, Windows Boot Manager, the EFI stub in the Linux kernel. Common ways to modify boot entries: the firmware setup (aka "the BIOS"), Linux `efibootmgr`, and `bcfg` in the UEFI Shell.
2. BCD entries are stored in the BCD and contain entries for the Windows Boot Manager, which is just one of many different types of boot managers you can choose from in (1). Common ways to modify BCD entries: `bcdedit.exe`.

It looks like you, or some software you used, have confused (1) and (2), which for example is why there are "Windows Boot Manager" items in your Windows boot selection menu, which itself is already being presented by the Windows Boot Manager.

Use one of the tools I mentioned (or something else - there are more than enough options here) to fix your system so that it at least boots normally. Then, if **after** installing EfiGuard, you run into some issue that is **related** to EfiGuard... create a new issue.

closed 2023-10-03 by Mattiwatti

MrTatiTunes, 2023-10-04:
I don't know what is wrong with UEFI, but it does not work for my PC.
I have ghost options in BIOS start screen.
Furthermore, I have changed to BCD and still have them.
PC starts and functioning on all systems, so no big deal.
Will give some sample of behaviour sometime later.
I am a bit too busy to do it now.