Mattiwatti/EfiGuard #98: Disable DSE without disabling PG?

utoni, 2023-08-17:
My assumption: It is most likely not possible as I guess that PG checks the integrity of the code which prevents loading unsigned drivers and is overwritten/hooked by EfiGuard. Am I right?

utoni, 2023-08-17:
Nvm, it works..

Mattiwatti, 2023-08-22:
Hi. Yes, just to confirm, this should work: DSE exists to verify the integrity and authenticity of kernel drivers, whereas PatchGuard exists to verify the integrity of the kernel itself.

EfiGuard disabling/being able to disable DSE is more or less just a convenient consequence of it disabling PatchGuard, which is the primary purpose of EfiGuard. So while it's not possible to do this as-is (essentially because I designed it that way), you should be able to disable only DSE **at boot** (`EFIGUARD_DSE_BYPASS_TYPE` = `DSE_DISABLE_AT_BOOT`) by simply removing the call to `DisablePatchGuard` in the source code.

Disabling DSE **at runtime** without disabling PatchGuard will eventually trigger PatchGuard, since DSE is part of the kernel.

utoni, 2023-08-22:
> Disabling DSE **at runtime** without disabling PatchGuard will eventually trigger PatchGuard, since DSE is part of the kernel.

Maybe my approach needs some more testing. For now, I am just `#ifdef`'ed everything related to `DisablePatchGuard`, but still using the runtime DSE disable w/ `ci_opts`. I did not encounter any BSoD's for now.

Mattiwatti, 2023-08-23:
I would expect this to trigger a bugcheck eventually (give it some time - it can take up to a few hours sometimes). But I have to say I've never tried this with EfiGuard. This is just my speculation based on the fact that the original DSEFix was also incompatible with PatchGuard starting from Windows 8.1. See [PatchGuard incompatibility](https://github.com/hfiref0x/DSEFix#patchguard-incompatibility) in the DSEFix README.

ValeZAA, 2024-08-21:
> This is just my speculation based on the fact that the original DSEFix was also incompatible with PatchGuard starting from Windows 8.1. See [PatchGuard incompatibility](https://github.com/hfiref0x/DSEFix#patchguard-incompatibility) in the DSEFix README.

You are correct. I used DSE runtime disabler through vuln. dribers and loaded modified and thus invalidly signed driver, and .sys eventually bluescreened in CI.DLL... Your bootkit helps with that and it no longer bkuescreen. Thank you.