Mattiwatti/EfiGuard #102: Is there any way to disable PG at runtime?

qq1091192337, 2023-08-31:
Hello, I noticed that there are two functions in your project, DisableDSE and disabledPatchGuard, which are installed before loading the system. And there is a flag in DisableDSE that can control startup. So I wonder if PG can set a flag to turn off and turn on PatchGuard through the same user mode program operation. I don't know if my idea is feasible.

DecoderCoder, 2023-08-31:
afaik you can disable patchguard in runtime with exploit (there is no legit turn on/off function). Once you started it, it can't be turned off (if you do the same hook as DSE in PG function, PG will detect it in itself and throw bsod)

Mattiwatti, 2023-10-03:
Sorry, no. There is no single global flags field that will disable or pause PatchGuard once it has been initialized, unlike DSE which can be controlled at runtime via `g_CiOptions`.

Runtime PatchGuard bypasses do exist: for example, see [ByePg](https://github.com/can1357/ByePg) (note: I'm not sure it will work with current versions of Windows).

EfiGuard, being a bootkit, is not well suited for doing something like this because - apart from being very hard to do to begin with - this requires either access to the NT kernel API (which EfiGuard does not have) for the required primitives, or else collecting very many runtime-variable addresses and offsets, which is error prone and difficult to do correctly even on one kernel, let alone across multiple versions of NT.

closed 2023-10-03 by Mattiwatti