Mattiwatti/EfiGuard #105: macbook white 13" late 2009

Padke9, 2023-09-15:
----Using the UEFI shell to load the driver-----

downloaded the [EDK2 UEFI Shell](https://github.com/tianocore/edk2/blob/edk2-stable201903/ShellBinPkg/UefiShell/X64/Shell.efi?raw=true) and rename it to bootx64.efi.
Boot the machine to the UEFI shell.

[image]
[image]

stucked in this place 

or if i  use Booting the loader

Download  EfiGuard, go to EFI/Boot and rename one of Loader.efi or Loader.config.efi to bootx64.efi. 

Place the files on a boot drive   /EFI/Boot/EfiGuardDxe.efi   and  /EFI/Boot/bootx64.efi

will stuck in white screen

Mattiwatti, 2023-10-01:
I'm almost certain this is some kind of incompatibility caused by the very old version of UEFI, or probably even EFI 1.10, supported by your MacBook's firmware. (I am assuming you've already updated it.)

From your photo it can be seen that there are already issues with printing the path to a loaded image. Of course this is not critical functionality, but there are several places in the EfiGuard DXE driver where the device path API **is** providing critical functionality. And this is just one possible issue that comes to mind due to the obvious missing path in your photo - there could be others.

Unfortunately it is very hard for me to say what the issue(s) is/are precisely (and whether they are fixable) without having access to such a machine myself, so I will leave this open for now. If anyone wants to send me a MacBook from around 2009 or so I'll be happy to look into this.

Padke9, 2023-10-05:
yes this is certainly  incompatibility issue due  old version of UEFI and there is no new MacBook's firmware updates available .
so i cannot use above EFI 1.10 on this laptop.  Is there any workaround for this or just forgot about this laptop.

Mattiwatti, 2023-10-05:
I'm pretty sure that this is something that would have to be worked around in the EfiGuard source code in some way to get this to work. Which I'd be willing to do, but as I don't have a MacBook to debug this it's very hard for me to say what the issue is.

closed 2025-06-13 by Padke9