Mattiwatti/EfiGuard #107: Backdooring bootmgfw.efi

gabriel-maxx, 2023-10-14:
### Question description

Hello, I would like to know what I should modify in the project to do this?

for example, I compile the project and it gives me an .efi driver and then I rename this driver to bootmgfw.efi and then I run it in cmd: mountvol S: /S

I delete the original and put bootmgfw.efi in S:\EFI\Microsoft\Boot
and when I restart windows, everything will be automatically disabled without messages etc..

Mattiwatti, 2023-10-14:
Your question seems very confused to me. EfiGuard doesn't replace the Windows Boot Manager - in fact, it depends on it. So the answer to your question would be something like:
1. Recreate the Windows Boot Manager from scratch in C, then hit compile.

I'm guessing this isn't what you mean, so please clarify. What is it that you need EfiGuard to do, that the source code you compiled does not do?

gabriel-maxx, 2023-10-15:
> Your question seems very confused to me. EfiGuard doesn't replace the Windows Boot Manager - in fact, it depends on it. So the answer to your question would be something like:
>
> 1. Recreate the [...]

What I want is to add a command to the program that basically specifies a Windows bootloader to infect, for example: EfiDSEFix.exe -f bootmgfw.efi --output backdored_bootmgfw.efi

and then I copy it to S:\EFI\Microsoft\Boot
and when windows starts PatchGuard and Driver Signature Enforcement will be disabled, do you understand me now?

Mattiwatti, 2023-10-16:
Things you are confused about in your post:

1. EfiGuard doesn't 'infect' anything as it works entirely at runtime and in memory. (Also, hopefully with the consent of the user... 'infecting' is something malware does.)
2. `Bootmgfw.efi` is not the bootloader, it is the boot manager. `winload.efi` is the bootloader. Depending on your reasons for wanting to do this (which aren't clear to me), you may need to patch one, the other, or both.
3. `EfiDSEFix.exe` is not EfiGuard. EfiDSEFix is only a very basic Windows program provided for convenient DSE overriding, which is the most obvious (but far from the only possible) thing one might want to talk to the driver for in Windows. EfiDSEFix doesn't even know how to save a file to disk, let alone patch the boot manager.
4. "Windows starts PatchGuard and Driver Signature Enforcement will be disabled" **is already what EfiGuard does** - but without the bizarre added requirement of having to replace the Windows boot manager. Not wanting to do this is the entire reason EfiGuard is a bootkit in the first place.

Not doing this because of (4). You may want to take a look at [UPGDSED](https://github.com/hfiref0x/UPGDSED), the original inspiration for EfiGuard, which does patch files on disk and sounds closer to what you (think you) need.

closed 2023-10-16 by Mattiwatti