Mattiwatti/EfiGuard #108: I have an error with EfiDSEFix.exe -d

nguyenquy96, 2023-10-20:
### Question description

When i use EfiDSEFix.exe -d then throw error

Disabling DSE...
CI!g_CiOptions at 0xFFFFF80168F13004.
Failure: NtSetSystemEnvironmentValueEx error 0xC0000022
The EfiGuard DXE driver is either not loaded in SETVARIABLE_HOOK mode, or it is malfunctioning.
AdjustCiOptions failed: 0xC0000022

infor EfiDSEFix.exe -i:
SystemBootEnvironmentInformation:
        - BootIdentifier: {4bb93527-e1ec-11ed-a2df-e083304d0408}
        - FirmwareType: UEFI
        - BootFlags: 0x0

SystemModuleInformation:
        - Kernel: ntoskrnl.exe (\SystemRoot\system32\ntoskrnl.exe)

SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x0001
           0x0001: CODEINTEGRITY_OPTION_ENABLED

SystemKernelDebuggerInformation:
        - KernelDebuggerEnabled: 0
        - KernelDebuggerNotPresent: 1

SystemKernelDebuggerInformationEx:
        - DebuggerAllowed: 0
        - DebuggerEnabled: 0
        - DebuggerPresent: 0

SharedUserData->KdDebuggerEnabled: 0x00

SystemKernelDebuggerFlags: 0x00

SystemCodeIntegrityPolicyInformation:
        - Options: 0x40000000
        - HVCIOptions: 0x0000

SystemIsolatedUserModeInformation:
        - SecureKernelRunning: 0
        - HvciEnabled: 0
        - HvciStrictMode: 0
        - DebugEnabled: 0
        - FirmwarePageProtection: 1
        - EncryptionKeyAvailable: 0
        - TrustletRunning: 0
        - HvciDisableAllowed: 0
** my system support UEFI && GPT is avaliable, please help

Mattiwatti, 2023-10-20:
I'm *guessing* you did not load EfiGuard during boot, i.e. what the error message is already telling you. However, in that case I would expect `EfiDSEFix.exe` to report NTSTATUS `0xC000000D` (`STATUS_INVALID_PARAMETER`), not `0xC0000022` (`STATUS_ACCESS_DENIED`) like you are seeing.
So, just to confirm... you did start the system with EfiGuard loaded (see: [README.md#Usage](https://github.com/Mattiwatti/EfiGuard/blob/master/README.md#usage))?

If yes, I'll need some more information:
- What version of Windows are you using?
- Are you running as administrator?
- What motherboard are you using, and with what BIOS version? (or if this is a VM, what VM software is this?)

nguyenquy96, 2023-10-23:
Yes ! Thank for help. I was try during boot then it's worked. Thank you !

closed 2023-10-23 by Mattiwatti