Mattiwatti/EfiGuard #109: it's me again lol

Fraudly, 2023-11-07:
### Operating system

Windows 10

### Issue description

#
Hi again, haven't posted in a while, just forgot about it, but after a while I tried to run efi again, even bought a new flash drive and with rufus made the partition as GPT, but I didn't understand why the installation didn't
go at all, there wasn't even any error at all.

### Steps to reproduce

1. 
2. 

### Attachments

[image]

niansa, 2023-12-01:
As the error explains, you need your Windows installation to be UEFI, **not** legacy.

acoul, 2024-06-12:
I guess this ticket may now rest in the closed ones :)

closed 2025-04-18 by Mattiwatti