Mattiwatti/EfiGuard #110: Error Boot

SatanJs7Discord, 2023-11-07:
### Operating system

Windows 10 x64

### Issue description

The problem is that when I boot my computer with EfiGuar, in the bios I move my usb in 1st place, after my ssd and at the end my hdd. But idk why when I launch my windows boot on my hdd files.

### Steps to reproduce

1. 
2.

acoul, 2024-01-04:
>  But idk why when I launch my windows boot on my hdd files.

did you solved your issue ? if not, can you try to use a better translator and discribe better your problem ?

Mattiwatti, 2024-01-07:
Closing this. Your post is borderline unintelligible. It is also missing any kind of relevant information whatsoever that would allow me to tell you what you're doing wrong. I don't know what's on your USB drive, I don't know what's on your SSD, and I don't know what's on your HDD. *How can you expect me to point out the cause of the issue for you?*

The **only** thing I'm quite certain of is that this (whatever 'this' is...) is not a bug in EfiGuard, but rather some misunderstanding or failure to follow instructions on your end.

Feel free to reopen this to add **detailed, readable** reproduction steps if you believe I'm in the wrong here and that this really is a bug.

closed 2024-01-07 by Mattiwatti