Mattiwatti/EfiGuard #116: Legacy (non_UEFI) Windows Installations

SulZo, 2024-03-16:
### Operating system

Windows 10 x64 22H2

### Issue description

[image]
What OS should I have to fix it? or I don't need to reinstall new OS?
Using the latest 1.4 version

### Steps to reproduce

1. 
2.

Mattiwatti, 2024-03-20:
Your Windows *version* probably isn't the problem, unless this is Windows XP or older.

Rather, the issue is that Windows is installed in 'legacy' (BIOS/MBR) mode. Reinstalling Windows is one option, but you don't necessarily have to. [Microsoft MBR2GPT](https://learn.microsoft.com/en-us/windows/deployment/mbr-to-gpt) supports automatic conversion to from MBR to UEFI/GPT installations for Windows 10 and 11, and (although not officially supported) I've successfully used it to convert installations of older versions such as Windows 7 as well.

acoul, 2024-06-12:
FWIW, I am able to use efiguard on a UEFI-3 system on win8.1 installed on mbr

[image]

ibay770, 2024-07-15:
> Your Windows _version_ probably isn't the problem, unless this is Windows XP or older.
> 
> Rather, the issue is that Windows is installed in 'legacy' (BIOS/MBR) mode. Reinstalling Windows is one option, but you don't necessarily have to. [Microsoft MBR2GPT](https://learn.microsoft.com/en-us/windows/deployment/mbr-to-gpt) supports automatic conversion from MBR to UEFI/GPT installations for Windows 10 and 11, and (although not officially supported) I've successfully used it to convert installations of older versions such as Windows 7 as well.

I"ve been wondering, is there a way to extract the patched files efiguard processed and replace them instead of the originals? This way, it may be possible to use MBR too, and efiguard won't be needed.