Mattiwatti/EfiGuard #119: Source can't compile on latest EDK because gEfiLegacyBiosProtocolGuid removed

nProtect, 2024-04-19:
### Question description

Hello,

Seem source code can't compile if use with latest EDK because gEfiLegacyBiosProtocolGuid get removed.
Ref: https://edk2.groups.io/g/devel/topic/102518665

To temporary fix this, I need to comment this code to make it can compile on latest EDK

`EFI_LEGACY_BIOS_PROTOCOL *LegacyBios;
Status = gBS->LocateProtocol(&gEfiLegacyBiosProtocolGuid,
	NULL,
	(VOID**)&LegacyBios);
ASSERT_EFI_ERROR(Status);

BootOptions[Index].Status = LegacyBios->LegacyBoot(LegacyBios,
	(BBS_BBS_DEVICE_PATH*)BootOptions[Index].FilePath,
	BootOptions[Index].OptionalDataSize,
	BootOptions[Index].OptionalData);`

Is there any issue if I remove this code?

Mattiwatti, 2024-04-22:
Indeed, EDK2 has removed all CSM support from OVMF. I actually ran into this issue a while ago, and I have fixed this locally by simply copying the minimum amount of code necessary from OVMF so that the same code in EfiGuard's loader can continue to compile (as the protocol and its GUID are actually used in 'real' PC firmware, not just OVMF, and I do not want to remove the code, as it prevents people from mistakenly thinking EfiGuard is not working when the real issue is the fact that their Windows is installed in BIOS mode.)

My main issue at the moment is actually the fact that I don't wish to encourage compilation using EDK2 master in the first place, but for different reasons: see #112, which is a more serious bug in current edk2 master that I do not currently have a fix or workaround for. Nevertheless, I will probably commit and push a fix for this sometime in the next few days, assuming I find time to test it.

To answer your question:
> Is there any issue if I remove this code?

Nope. This code was only added as a way to help out idiots who are trying to use **Efi**Guard on a legacy (BIOS/MBR) installation of Windows, and point out to them why this is not going to work. Assuming you are using EfiGuard on an EFI installation of Windows as intended, this code will never be executed, and it is safe to remove it.

nProtect, 2024-04-23:
> Indeed, EDK2 has removed all CSM support from OVMF. I actually ran into this issue a while ago, and I have fixed this locally by simply copying the minimum amount of code necessary from OVMF so that [...]

Thank you. I only modify some to use only myself and my windows installed as UEFI.

Mattiwatti, 2025-01-17:
This is now fixed. Sorry for the wait!

commit 5bfa808f (2025-01-17): Fix build with current EDK2 master

Re-add the missing headers for EFI_LEGACY_BIOS_PROTOCOL and gEfiLegacyBiosGuid which were removed from OvmfPkg in https://github.com/tianocore/edk2/commit/9d4becddbad24ca00b01d68087e0e86d33ec92b6.

Neither OVMF nor EfiGuard actually *needs* the CSM for anything, but being able to check for the presence of the legacy BIOS protocol is still important in order to determine if we are actually booting a legacy BIOS boot option. This allows us to then inform the user that what they are doing is not suppported and most likely a mistake.

Fixes #119
closed 2025-01-17 by Mattiwatti