Mattiwatti/EfiGuard #121: Files patched but unsigned driver not work

Gelip, 2024-05-15:
### Operating system

Windows 7 SP1 build 17514

### Issue description

Patched files successfully but the unsigned driver BeepXP does not work:
Cannot open Beep driver: 2

### Steps to reproduce

I try both method - loader and driver. The application EfiDSEFix sees hook.

### Logs

```shell
SystemBootEnvironmentInformation:
        - BootIdentifier: {453c784e-e38f-11ee-a703-cdd306f600bc}
        - FirmwareType: UEFI
        - BootFlags: 0x0

SystemModuleInformation:
        - Kernel: ntoskrnl.exe (\SystemRoot\system32\ntoskrnl.exe)

SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x0001
           0x0001: CODEINTEGRITY_OPTION_ENABLED

SystemKernelDebuggerInformation:
        - KernelDebuggerEnabled: 0
        - KernelDebuggerNotPresent: 1

SharedUserData->KdDebuggerEnabled: 0x00

efidsefix -c
Checking for working EFI SetVariable hook...
Success.

efidsefix -r
Querying g_CiEnabled value...
nt!g_CiEnabled at 0xFFFFF80001C7CEB8.
Success. g_CiEnabled value: 0x0
```

Mattiwatti, 2024-05-17:
You do understand that in order for me to reproduce your issue, I will (at the very least) need *both*:
1. Your "beepxp driver", as well as precise steps on how to install it (meaning **exactly** the steps **you** used to install this driver on **your** system), and
2. The actual file(s) for your "no_pg_ds_v3" fix, as I do not have an account on exetools.com, nor would I even know what file(s) to download if I did, or what to do with them in order to recreate **your** exact setup.

Basically: I cannot read your mind, and I also do not have remote access to the contents of your hard drive.

So, assuming you want me to reproduce this so that I can fix it (and also assuming there is something to fix in EfiGuard here), I'm going to need a lot more information than this.

Gelip, 2024-05-18:
BeepXP driver > [http://www.waldbauer.com/tmp/dl.php?download=beepxp64](http://www.waldbauer.com/tmp/dl.php?download=beepxp64)
Install instruction:

1) Disable Windows' own BEEP.SYS driver

        sc config Beep start= demand

2) Right click onto the beepxp.inf file and choose "Install."
   The driver will now be installed into the system. If there are no errors
   on installation, it will silently proceed.

3) Reboot your machine so that the driver gets loaded on startup.

Mattiwatti, 2024-05-23:
Not only does the driver you linked work fine on my Windows 7 installation, you don't even need EfiGuard or test signing enabled to load this driver at all since it has a valid SHA256 signature (as stated in the README...).

My only guess is that you are missing the SHA256 signing updates [KB4474419](https://support.microsoft.com/help/4474419) and [KB4490628](https://support.microsoft.com/help/4490628) for Windows 7. You should have these installed anyway as they have been required for Windows 7 updates since 2019.

If for some reason you must use an unpatched Windows 7, you need to configure EfiGuard's loader to disable DSE at boot time instead of disabling it at runtime. This is required for this driver as it is configured as system start, meaning the driver has already failed to load by the time you are disabling DSE.

Closing this as this is not a bug in EfiGuard.

closed 2024-05-23 by Mattiwatti