Mattiwatti/EfiGuard #132: Patched ntoskrnl fails to load with STATUS_IMAGE_CHECKSUM_MISMATCH.

Hallilogod, 2024-09-06:
### Operating system

Windows 10 build 19045

### Issue description

I am trying to load a custom ntoskrnl image. EfiGuard does not give any errors during its log and reports functions like ImgpValidateImageHash or ImgpFilterValidationFailure as successfully patched, however winload reports the image as corrupted with ntstatus 0xC0000221 (STATUS_IMAGE_CHECKSUM_MISMATCH). I am doing this on VMware, which should not cause any problems.

### Steps to reproduce

1. Replace ntoskrnl.exe on a Windows 10 19045 system with a custom image you want to load.
2. Boot the EfiGuard loader.

### Attachments

[image]
[image]

aubymori, 2024-10-04:
@Hallilogod Try loading with boot time DSE bypass by pressing Home when it prompts you and selecting the option for boot time DSE bypass.

Mattiwatti, 2025-06-11:
Getting this to work was a bit more work than I expected. This was mostly because apart from kernel images, I also wanted this to work for boot managers and boot loaders, which in turn exposed a nasty bug that I've only just committed a fix for (6cb7f85).

If there are no unexpected regressions due to the above changes (I don't expect there will be), then I'll commit and push a fix for this issue soon. With the upcoming change, EfiGuard should now be able to let Windows boot an entire chain of images with invalid checksums:
[image]

Note: I don't consider this issue to be of particularly high importance, given that fixing the checksum for a PE image is trivial. However, I also don't really have any objections to it so long as the change does not introduce regressions or needlessly complicate the codebase, which doesn't seem to be an issue at the moment.

Hallilogod, 2025-07-06:
Thanks for looking into this after such a long time! I didn't actually know that the image checksum mismatch error is caused by the PE's Checksum mismatch and thought the checksum of the legitimate kernel would be hardcoded, definitely good to know.

commit 801ad433 (2025-08-03): PatchImgpValidateImageHash: help out images with invalid PE checksums

Fixes #132
closed 2025-08-03 by Mattiwatti